emtrax-ltd/Cura2MoonrakerPlugin #1: Cura Crash

swalsh11, 2021-02-25:
This is my first test trying to get this to run,  as soon as I attempt to send a g code file, Cura falls over with the error log below: 
Cura 4.8.0, RPi 4,  with new install  Fluidd, Klipper, Kiauh, Klipperscreen,  Spi 5"touchscreen,  

Traceback (most recent call last):
  File "C:\Users\swalsh11\AppData\Roaming\cura\4.8\plugins\Cura2MoonrakerPlugin-main\MoonrakerPlugin\MoonrakerOutputDevice.py", line 143, in onFilenameAccepted
    self._sendRequest('printer/info', on_success = self.onInstanceOnline)
  File "C:\Users\swalsh11\AppData\Roaming\cura\4.8\plugins\Cura2MoonrakerPlugin-main\MoonrakerPlugin\MoonrakerOutputDevice.py", line 237, in _sendRequest
    headers['Authorization'] = 'Basic ' + base64.b64encode(auth)
TypeError: Can't convert 'bytes' object to str implicitly

Log:

Thread 0x0000922c (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00008658 (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001f14 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1262 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00008e2c (most recent call first):
  File "zeroconf\__init__.py", line 1183 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000094b0 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000067e0 (most recent call first):
  File "serial\tools\list_ports_windows.py", line 291 in iterate_comports
  File "serial\tools\list_ports_windows.py", line 299 in comports
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 118 in getSerialPortList
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 85 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00009ba8 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000476c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

[... 73 lines cut ...]
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x000024d0 (most recent call first):
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 396 in _logInfoWidget
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 182 in _createDialog
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 86 in __init__
  File "X:\4.8-exe\build\inst\bin\cura_app.py", line 167 in exceptHook
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 378 in exec_
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 847 in run
  File "X:\4.8-exe\build\inst\bin\cura_app.py", line 235 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

emtrax-ltd, 2021-02-26:
i commited the fix for this issue into the repo  - give it a try

swalsh11, 2021-02-26:
That Resolved the Issue,  Thank you .

closed 2021-02-26 by swalsh11