emtrax-ltd/Cura2MoonrakerPlugin #2: [FR] replacing dots and whitespace in filenames with an underscore or dash

itCarl, 2021-03-22:
it would be really nice if the Plug-in could replace dots and whitespaces/spaces with an underscore or a dash instead of cutting/stripping the Filename.

**Examples**:
i have a .stl file with the filename "_test_head_v1.4 (small).stl_"
current filename handling is like "_test_head_v1.[gcode | ufp]_" (file extension doesn't really matter in thsi example)

would i would like is something like this "_test_head_v1_4_small.gcode_" or "_test_head_v1_4-small.gcode_"

emtrax-ltd, 2021-03-22:
there is currently no cutting or stripping algorithm for filenames implemented - it takes the suggestion from cura (core) for the filenames to use -> same as for dialog of "save as file". is there any other plugin active in your installation of cura which can enforce this behavior?

no matter... i can imagine a feature for regexp-replace in the filename - so you can 'program' and configure the conversion for your self.

itCarl, 2021-03-23:
> is there any other plugin active in your installation of cura which can enforce this behavior?

currently i dont have any plug-ins except "yours" (Cura2MoonrakerPlugin)

But Fenne from Ultimaker/Cura#9431 linked me a Plug-in that exactly does what this issue here is about.
in general, i don't like plug ins, i prefer fixed features because then less can go wrong more often

emtrax-ltd, 2021-03-23:
in the upload-dialog there is only one limitation for filenames ->  maximumLength: 100; 

if you need more, just post it.

itCarl, 2021-03-23:
i just played around with the Plug-in and stuff and just found something interesting...

> use -> same as for dialog of "save as file".

**Save to File**
[image]

**Upload to Moonraker**
[image]

[image]
[image]

emtrax-ltd, 2021-03-23:
I will check that

itCarl, 2021-03-23:
I'm trying to figure out were exactly the 'error'
here is what i found so far.

---

if I am not mistaken it's doing the "os.path.splitext(file)" correctly..  (line 94)
...thats weird but its displaying it wrong in the dialog box...

[image]

https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/blob/97043a926376cabc6882f93832f9e6bc52678b25/MoonrakerPlugin/MoonrakerOutputDevice.py#L92-L107

---
**Just some quick and dirty Debuging** 
[image]
[image]

emtrax-ltd, 2021-03-23:
thank you for your detailed investigation. i have taken it into account and developed it a little bit more complex but much more flexible. take a look at the new version in the repository.

```
       if fileName:
            fileName = os.path.basename(fileName);
        else:
            fileName = "%s." % Application.getInstance().getPrintInformation().jobName
        
        # Translate filename
        if self._trans_input and self._trans_output:
            transFileName = fileName.translate(fileName.maketrans(self._trans_input, self._trans_output, self._trans_remove if self._trans_remove else ""))
            fileName = transFileName

        self._fileName = fileName  + "." + self._output_format
```

your requirements should be satisfied by the following settings... now available within the configuration wizzard
[image]

itCarl, 2021-03-23:
That's cool.
Thanks a lot.

closed 2021-03-23 by itCarl