emtrax-ltd/Cura2MoonrakerPlugin #4: Display not large. And port number is required for connect to printer.

rexstor, 2021-03-25:
Using CURA 4.8
Fluidd with Klipper

For the display
[image]
The button cropped. but i already adjust 
actionDialog.minimumHeight = screenScaleFactor * 320;
and it seams fit nicely.
[image]

IP address need to put including port number.
If only IP address without port, the file cant be uploaded.

http://192.168.9.112:7125/

Anyway thanks for great plug-in!

emtrax-ltd, 2021-03-26:
The port must be specified if it deviates from the standard http = 80 or https = 443 - this is practically enforced by the http protocol and headers.

emtrax-ltd, 2021-03-26:
question: isn't your moonraker behind a nginx?

closed 2021-04-05 by emtrax-ltd

GHswitt, 2021-04-12:
@emtrax-ltd I needed to change that to 360, otherwise I couldn't see the Save/remove config buttons.

emtrax-ltd, 2021-04-15:
@GHswitt fixed in #5