emtrax-ltd/Cura2MoonrakerPlugin #14: Doesn't work with the popular smartavionics fork of Cura

m-sundman, 2021-08-08:
Smartavionics makes a popular fork tracking the newest Cura release, and for some reason this plugin doesn't work with it.

Looks like some python version compatibility issue, but perhaps it could be circumvented.

```2021-08-04 16:28:33,992 - ERROR - [MainThread] UM.Logger.logException [113]: Exception: Import error loading module MoonrakerPlugin
2021-08-04 16:28:34,010 - ERROR - [MainThread] UM.Logger.logException [117]: Traceback (most recent call last):
2021-08-04 16:28:34,045 - ERROR - [MainThread] UM.Logger.logException [117]:   File "/home/markb/3dp/cura/cura-build.git/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py", line 681, in _findPlugin
2021-08-04 16:28:34,075 - ERROR - [MainThread] UM.Logger.logException [117]:   File "/opt/cura/lib/python3.8/imp.py", line 244, in load_module
2021-08-04 16:28:34,087 - ERROR - [MainThread] UM.Logger.logException [117]:   File "/opt/cura/lib/python3.8/imp.py", line 216, in load_package
2021-08-04 16:28:34,104 - ERROR - [MainThread] UM.Logger.logException [117]:   File "<frozen importlib._bootstrap>", line 702, in _load
2021-08-04 16:28:34,119 - ERROR - [MainThread] UM.Logger.logException [117]:   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
2021-08-04 16:28:34,131 - ERROR - [MainThread] UM.Logger.logException [117]:   File "<frozen importlib._bootstrap_external>", line 783, in exec_module
2021-08-04 16:28:34,141 - ERROR - [MainThread] UM.Logger.logException [117]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
2021-08-04 16:28:34,150 - ERROR - [MainThread] UM.Logger.logException [117]:   File "/home/marcus/.local/share/cura/master/plugins/MoonrakerPlugin/MoonrakerPlugin/__init__.py", line 6, in <module>
2021-08-04 16:28:34,159 - ERROR - [MainThread] UM.Logger.logException [117]:     from . import MoonrakerPlugin, MoonrakerAction
2021-08-04 16:28:34,168 - ERROR - [MainThread] UM.Logger.logException [117]:   File "/home/marcus/.local/share/cura/master/plugins/MoonrakerPlugin/MoonrakerPlugin/MoonrakerAction.py", line 5, in <module>
2021-08-04 16:28:34,178 - ERROR - [MainThread] UM.Logger.logException [117]:     from typing_extensions import ParamSpecKwargs
2021-08-04 16:28:34,187 - ERROR - [MainThread] UM.Logger.logException [117]: ImportError: cannot import name 'ParamSpecKwargs' from 'typing_extensions' (/tmp/.mount_Cura-mUEZ1HW/usr/bin/lib/library.zip/typing_extensions.pyc)
2021-08-04 16:28:34,203 - WARNING - [MainThread] UM.PluginRegistry._populateMetaData [773]: Could not find plugin MoonrakerPlugin
2021-08-04 16:28:34,216 - ERROR - [MainThread] UM.Logger.logException [113]: Exception: Import error loading module MoonrakerPlugin
2021-08-04 16:28:34,231 - ERROR - [MainThread] UM.Logger.logException [117]: Traceback (most recent call last):
2021-08-04 16:28:34,245 - ERROR - [MainThread] UM.Logger.logException [117]:   File "/home/markb/3dp/cura/cura-build.git/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py", line 681, in _findPlugin
2021-08-04 16:28:34,258 - ERROR - [MainThread] UM.Logger.logException [117]:   File "/opt/cura/lib/python3.8/imp.py", line 244, in load_module
2021-08-04 16:28:34,289 - ERROR - [MainThread] UM.Logger.logException [117]:   File "/opt/cura/lib/python3.8/imp.py", line 216, in load_package
2021-08-04 16:28:34,303 - ERROR - [MainThread] UM.Logger.logException [117]:   File "<frozen importlib._bootstrap>", line 702, in _load
2021-08-04 16:28:34,321 - ERROR - [MainThread] UM.Logger.logException [117]:   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
2021-08-04 16:28:34,342 - ERROR - [MainThread] UM.Logger.logException [117]:   File "<frozen importlib._bootstrap_external>", line 783, in exec_module
2021-08-04 16:28:34,359 - ERROR - [MainThread] UM.Logger.logException [117]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
2021-08-04 16:28:34,379 - ERROR - [MainThread] UM.Logger.logException [117]:   File "/home/marcus/.local/share/cura/master/plugins/MoonrakerPlugin/MoonrakerPlugin/__init__.py", line 6, in <module>
2021-08-04 16:28:34,397 - ERROR - [MainThread] UM.Logger.logException [117]:     from . import MoonrakerPlugin, MoonrakerAction
2021-08-04 16:28:34,422 - ERROR - [MainThread] UM.Logger.logException [117]:   File "/home/marcus/.local/share/cura/master/plugins/MoonrakerPlugin/MoonrakerPlugin/MoonrakerAction.py", line 5, in <module>
2021-08-04 16:28:34,439 - ERROR - [MainThread] UM.Logger.logException [117]:     from typing_extensions import ParamSpecKwargs
2021-08-04 16:28:34,462 - ERROR - [MainThread] UM.Logger.logException [117]: ImportError: cannot import name 'ParamSpecKwargs' from 'typing_extensions' (/tmp/.mount_Cura-mUEZ1HW/usr/bin/lib/library.zip/typing_extensions.pyc)
```

emtrax-ltd, 2021-08-09:
try to uncomment line 5 in MoonrakerAction.py

#from typing_extensions import ParamSpecKwargs

is the problem fixed now?

m-sundman, 2021-08-09:
Yes, now it works! Great! At least with this fork, but I haven't tried if it works with the mainline cura now.

emtrax-ltd, 2021-08-09:
the fix also works with mainline cura

closed 2021-08-09 by emtrax-ltd