emtrax-ltd/Cura2MoonrakerPlugin #16: Unable to save configured plugin settings

agrohn, 2021-08-09:
Unable to save configured plugin settings

Despite typing text in all fields, "Connect Moonraker" window shows only "Close" button, and no "Save Config" button at all. 
In Cura, printing buttons show "Moonraker Plugin", and clicking it displays message to Configure Moonraker in Preferences. 

System information: 

Gentoo Linux
Python 3.8
Cura 4.9.1

emtrax-ltd, 2021-08-09:
can you provide screenshots?

agrohn, 2021-08-09:
I was just going over QML and got around the issue by defining minimum height in `resources/qml/MoonrakerConfiguration.qml` as follows: 

```
    Component.onCompleted: {
        actionDialog.minimumWidth = screenScaleFactor * 580;
        actionDialog.minimumHeight = screenScaleFactor * 370;
        actionDialog.maximumWidth = screenScaleFactor * 580;
        actionDialog.maximumHeight = screenScaleFactor * 370;
        minimumHeight = 500;
    }
```
But this was a quick and dirty solution. Somehow the height of buttons is not reflected to window itself.

agrohn, 2021-08-09:
And for the screenshots, there is the original window without modifications to QML:
[image]

And after explicitly defining minimum height, buttons appeared: 
[image]

emtrax-ltd, 2021-08-09:
i will change height to "410" - does it work for you?

    Component.onCompleted: {
        actionDialog.minimumWidth = screenScaleFactor * 580;
        actionDialog.minimumHeight = screenScaleFactor * 410;
        actionDialog.maximumWidth = screenScaleFactor * 580;
        actionDialog.maximumHeight = screenScaleFactor * 410;
    }

agrohn, 2021-08-09:
I can confirm that buttons are visible with those settings, thanks!

emtrax-ltd, 2021-08-09:
fix commited

closed 2021-08-09 by emtrax-ltd