emtrax-ltd/Cura2MoonrakerPlugin #19: Remember choice in upload dialog for "Start print job"

bitfabrikken, 2021-09-28:
Hi, thank you for the great plugin.

I'd like to press as few buttons as possible to upload to Moonraker + start the print (RSI).

Would it be possible to e.g. remember the check selection in the "Start print job" checkbox, or maybe even have an option for changing the blue upload button to "Upload+Print".

Also would be nice to be able to not have the "Print sent to ...." info box pop up after every upload :)

emtrax-ltd, 2021-10-06:
what about two buttons "Upload" and "Upload & Print " instead of the checkbox?

emtrax-ltd, 2021-10-20:
update: automatic hiding of "successful" messages (after 30 seconds) will be implemented in the next days.

emtrax-ltd, 2021-11-03:
FR implemented - checkout newest version

closed 2021-11-03 by emtrax-ltd