emtrax-ltd/Cura2MoonrakerPlugin #21: Upload the plugin to the marketplace.

nallath, 2021-11-09:
I think this would be much easier to find if this was uploaded to the marketplace! You can do that via https://contribute.ultimaker.com

Even if you don't want that, you can use it to generate .curapackage files which can be drag & dropped into Cura, making installing it a lot less error prone.

psiberfunk, 2021-12-14:
I agree, this would be much more useful on the marketplace

mvoss96, 2021-12-19:
yes please.
I really like using this plugin but because it isn't in the marketplace you have to reinstall the plugin every time you update cura

psiberfunk, 2022-04-09:
I'd like to, but it's pretty frowned on to take someone's code as your own without permission

emtrax-ltd, 2022-04-09:
request in progress now

psiberfunk, 2022-04-09:
Yay !

emtrax-ltd, 2022-05-03:
[image]

Done!

KoenVanduffel, 2022-05-03:
Found it, removed the manually installed version, installed from the market.  It even remembers the settings. Testing to follow.

closed 2022-05-04 by nallath