emtrax-ltd/Cura2MoonrakerPlugin #24: Bug caused by pop-up window size

syl2333, 2021-12-03:
[image]
As you can see, the pop-up window is too small for me to see the save option below. Hope you can fix this problem as soon as possible.

batista16, 2022-01-07:
[image]

bdog1, 2022-01-19:
As a temporary suggestion you could change your Screen Resolution and that should allow you to at least be able to click the Save Button. Then change Screen resolution back afterwards.

emtrax-ltd, 2022-01-19:
i'm currently working on a new version for this dialog

keiseven, 2022-01-20:
When you have finished filling out the necessary information, click "remove box", press "Tab key" once, and press "space key" once, and the behavior will be the same as if you had clicked save.

closed 2022-01-22 by emtrax-ltd

emtrax-ltd, 2022-01-22:
The two dialogs "Connect" and "Upload" have been redesigned. I hope it is correct now.

commit eea06a3c (2022-01-22): Redesigned dialogs and fixed #24
commit 601ccfd7 (2022-01-26): Add files via upload

fixed #24 and #27