emtrax-ltd/Cura2MoonrakerPlugin #28: "Could not find plugin UFPWriter" error when attempting to use "UFP with Thumbnail" option

scramsby, 2022-01-29:
Cura version: 4.13.1

When I select the "UFP with Thumbnail" option in the "Connect to Moonraker" settings, and then attempt to send a sliced model to my printer, it just results in an error pop-up at the bottom of the Cura window that says:
"Error

Could not find plugin UFPWriter"

Changing the Moonraker option to send GCODE instead makes it work just fine. Might be due to breaking changes in Cura 4.13 as seems to be implied by https://www.reddit.com/r/Cura/comments/s5i0bg/cura_413_fails_to_loaf_plugin_ufpwriter/.

emtrax-ltd, 2022-01-29:
4.13 windows is fine

scramsby, 2022-01-29:
Maybe a regression in 4.13.1 then? I'm also reproing on a Windows 64-bit Cura build. Please let me know if any other logs or anything would be helpful.

emtrax-ltd, 2022-01-29:
4.13.1 windows also working

can you post the cura.log?

scramsby, 2022-01-30:
Your prompt to do so led me to see this in the log:
2022-01-29 17:00:05,670 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [429]: Plugin [UFPWriter] has been disabled. Skip loading it.

Going into the Marketplace->Installed->Bundled Plugins list, I could see UFPWriter listed but it was disabled. Clicking the checkbox to re-enable it and restarting Cura fixed everything. I presume maybe I disabled this at some point to optimize start-up/memory/etc.. Sorry for the noise. Hopefully this issue log will help someone else in the future.

closed 2022-01-30 by scramsby