emtrax-ltd/Cura2MoonrakerPlugin #29: Plugin Not Showing In Cura

teslabol77, 2022-01-31:
After Installation following the directions, the plugin does not appear under the printer settings
[cura.log](https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/files/7967529/cura.log)
.

Nordmann1961, 2022-02-02:
Have a look here:
https://github.com/emtrax-ltd/Cura2MoonrakerPlugin

The user based installation path is C:\Users\<Your Username>\AppData\Roaming\cura\[version number]\plugins (no admin privileges needed)

I had the same problem, now it works :-)

teslabol77, 2022-02-02:
I have it running fine on my PC Windows 11 installation. I cannot get it to show on my Mac OS installation following the same folder path listed.

psiberfunk, 2022-04-07:
This is why it would be better to package this plugin properly in the Ultimaker marketplace for free.. I'm not sure why this hasn't been done.

emtrax-ltd, 2022-04-09:
solution via #21

closed 2022-04-30 by emtrax-ltd