emtrax-ltd/Cura2MoonrakerPlugin #33: 5.0 support

devolf92, 2022-04-21:
I'm really in love with this plugin! Thanks for a great job!
Recently Cura released 5.0 beta but looks like they brake something and plugin stopped to work. While loading Cura says that plugin need to be reinstalled

LMS0815, 2022-04-22:
You can find addtitonal details at: https://community.ultimaker.com/topic/40288-ultimaker-cura-50-beta/

One major change is that the GUI framework Qt was updated from version 5 to version 6, to support newer operating systems (and fix multiple bugs with the previous version of Qt, which was no longer updated). Unfortunately this means that most of the UI of Cura and its plugins has to be made compatible with this newer version of Qt. Because of that all plugins need to be updated and resubmitted to the Marketplace.

emtrax-ltd, 2022-04-22:
yo! big change but easy fix. Buuut... i will take this opportunity to do some code refactoring.

PhilBaz, 2022-04-22:
@emtrax-ltd 

Looking forward to it!!!

Yeah, all plug-ins need to be reworked so currently the plugin 'market place' is totally empty...

emtrax-ltd, 2022-04-22:
OK... take a look at branch "cura5-refactored" and give it a try! tested it with cura 4.13 and cura 5.0 beta :)

cause of massive refactoring there could be some new bugs now ;)

PhilBaz, 2022-04-23:
@emtrax-ltd Crash on open....

[image]

soleshoe, 2022-04-23:
plugin not showing in cura 5.0 beta 1 on macOS

I tried adding a "plugins" folder in Cura App then adding plugin in it, but no success.
I tried to add the plugin in .../Application Support/cura/5.0/plugins, but no success either.

devolf92, 2022-04-23:
[image]
 that what i got with refactored version. but when i restarted app everything worked fine :D

emtrax-ltd, 2022-04-23:
> [image] that what i got with refactored version after pressing upload button. but when i restarted app everything worked fine :D

in these cases please attach the cura.log

PhilBaz, 2022-04-23:
With the updated version Cura now opens and I can get into the moonraker dialogue.....But  1) I cant enter an address, whatever I do it just doesn't hold a value. Double checking, nether dose API or name dialogue hold a value 2) I cant select the 'format' radio button to chose UFP.

[image]

devolf92, 2022-04-23:
[done](https://pastebin.com/tFVLw69r)

emtrax-ltd, 2022-04-23:
> [done](https://pastebin.com/tFVLw69r)

hmmm.... Not Found (#404)

:(

devolf92, 2022-04-23:
gdrive my last hope :D i edited my previous post

emtrax-ltd, 2022-04-23:
> [image] that what i got with refactored version after pressing upload button. but when i restarted app everything worked fine :D

ok... i saw the log and this error only came up twice. All other times (then later) did the dialog work without problems?

PhilBaz, 2022-04-24:
All good here after the last round....Seems to be working fine. Cura 5 to MainsailOS UFP with thumb.

Thank you very much!!!

emtrax-ltd, 2022-04-24:
branch merged...we need more "volunteers" now to find bugs ;)

KoenVanduffel, 2022-04-26:
Tested the updated plugin in Cura 5.0, Win 10 works like a charm, ufp thumbnail is there. Only the power on of the printer I have not tested yet (configured it and nothing complains at least).

emtrax-ltd, 2022-04-28:
next updates and new features in branch [cura5-refactored-monitor](/emtrax-ltd/Cura2MoonrakerPlugin/tree/cura5-refactored-monitor)

PhilBaz, 2022-04-29:
Sweet...ill give er a go in the morning.

PhilBaz, 2022-04-29:
@emtrax-ltd 

Transfer with thumb is working fine... I see the new cam dialogue. What address does it want? Mine is plugged into the Pi running klipper. Not sure what the address is or how to find it. If you can give me a hint ill test it.

KoenVanduffel, 2022-04-29:
I haven't tested the Webcam function yet but the url you can find in the settings dialog. Default is URL: /webcam/?action=snapshot

PhilBaz, 2022-04-29:
using :    /webcam/?action=snapshot    or    /webcam/?action=stream

  The monitor dialogue doesn't show an image.   I got the two URLs above from Mainsail OS. 

But.... Im also pulling the image in Mainsail OS at the moment. So maybe that's the issue?

KoenVanduffel, 2022-04-29:
I guess you have to prepend the url, or ip

PhilBaz, 2022-04-29:
Okay......got it!

http://xxx.xxx.x.xxx/webcam/?action=stream

emtrax-ltd, 2022-04-29:
i will update the logic for webcam url in one of the next versions: absolute (http[s]://..) or relative (/....?..) urls will be supported!

emtrax-ltd, 2022-04-29:
why "absolute" at this moment: because i dont have any webcams, so i only tested it with an free/private mjpeg-stream somewhere from the internet ;)

PhilBaz, 2022-04-29:
Yeah, working great!!

Thank you for the excellent toys!!! I would say you should register the plugin with Cura market so more people can use this awesome app. But I imagine its a huge pain in the ass to jump through their hoops.

emtrax-ltd, 2022-04-29:
The cura marketplace process is still ongoing -> their review step is very slow!

emtrax-ltd, 2022-04-30:
hopefully that was the last fix for the branch [cura5-refactored-monitor](/emtrax-ltd/Cura2MoonrakerPlugin/tree/cura5-refactored-monitor) - if there are no more bugs, then i will merge this into main

emtrax-ltd, 2022-04-30:
I think everything is done here

closed 2022-04-30 by emtrax-ltd