emtrax-ltd/Cura2MoonrakerPlugin #40: Feature Request: select / browse directory on moonraker to upload the file into

msteele999, 2022-05-29:
I don't know if this is possible but I figured I'd ask:

- I use several different directories on my klipper / moonraker / fluidd install for gcode files to keep them separated. Is it possible to have the plugin allow navigation of the root upload directory to select a subdirectory to upload into?

Thank you

emtrax-ltd, 2022-05-31:
It is currently possible to prepend a path to the filename in the upload dialog. I will check if a browsable list is possible.

emtrax-ltd, 2022-06-01:
The client API is limited for this requirement...
  - simple implementation: list directories under the root of gcodes only 
  - more complex and difficult: list tree structure recursively

closed 2022-09-16 by emtrax-ltd