emtrax-ltd/Cura2MoonrakerPlugin #41: Feature request: Bypass file rename window

mujeebbasit, 2022-06-01:
First, thank you for this great plugin.  It has allowed me to completely switch away from Octoprint.  The one thing I never do it change the filename prior to printing.  Therefore, the file rename popup is waste of a click for me.  Could you please add a parameter to auto accept the rename windows?

Thank y ou

emtrax-ltd, 2022-06-01:
There is a small problem with "Fire and Forget" -> "Upload Only" or "Upload & Print"? I think we need a way to set either of these two options.

mujeebbasit, 2022-06-11:
Agree, it would be nice to have an option in the setting for that.  I only use "Fire and Forget" because I always need to make sure the printer bed a clear.  The settings options would allow people to change their minds in the future.  Again, thank you for a great plugin; it has been working well for months now.

emtrax-ltd, 2022-06-12:
Will be available in the next version (1.6.3+).

closed 2022-06-12 by emtrax-ltd