emtrax-ltd/Cura2MoonrakerPlugin #50: Upload timeout too short

mujeebbasit, 2022-07-15:
Hello,

Thank you again for the "Fire and Forget" feature.  Is it possible to adjust the time before error?  When I first send a file, the printer turns on via relay and the firmware loads.  That take longer than the current delay before an error is displayed, and load is cancelled.  This could either be longer or a configurable option.  Other than this minor inconvenience, the plugin has worked very well.  

Best,
Mujeeb

emtrax-ltd, 2022-07-17:
Are you using the power device configuration of this Cura2MoonrakerPlugin?

jdmz1, 2022-09-12:
I have the same issue.

emtrax-ltd, 2022-09-16:
give branch "retry-interval" a try > https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/tree/retry-interval

jdmz1, 2022-09-17:
> give branch "retry-interval" a try > https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/tree/retry-interval

I just tested it a couple times with 1 second interval and it seems to be working. Thanks!  Will do more testing later.

emtrax-ltd, 2022-09-28:
New release 1.6.4 submitted to Ultimaker Cura Marketplace - will be available after their approval.

closed 2022-09-28 by emtrax-ltd