emtrax-ltd/Cura2MoonrakerPlugin #52: Crash when file upload CURA 5.1

CrownFlash, 2022-07-30:
Hi, Cura crashed whenever i upload a file.

`Traceback (most recent call last):
  File "UM\TaskManagement\TaskManager.py", line 142, in event
    call_event.callFunction()
  File "UM\TaskManagement\TaskManager.py", line 33, in callFunction
    self._function(*self._args, **self._kwargs)
  File "C:\Users\Ludovic\AppData\Roaming\cura\5.1\plugins\MoonrakerConnection\MoonrakerConnection\MoonrakerOutputDevice.py", line 305, in _checkPrinterStatus
    status = response['result']['state']
TypeError: string indices must be integers`

Should be corrected by converting the reponse into int :)

emtrax-ltd, 2022-07-30:
Strange! Can you provide the response from "http://<moonraker-address:port>/printer/info" (called via browser)... just before uploading?

CrownFlash, 2022-07-30:
Im not much of a programer for now but i'm trying.

When i use your link into the brower it goes to a google search of it.

How i am supposed to open the thing you want?
For info i access my printer via this adress "http://fluiddpi.local/#/"

CrownFlash, 2022-07-30:
Oh sorry i just understand how i was supposed to use it.

`{"result": {"state_message": "Printer is ready", "klipper_path": "/home/pi/klipper", "config_file": "/home/pi/klipper_config/printer.cfg", "software_version": "v0.9.1-616-g28f60f7e", "hostname": "fluiddpi", "cpu_info": "4 core ARMv7 Processor rev 3 (v7l)", "state": "ready", "python_path": "/home/pi/klippy-env/bin/python", "log_file": "/home/pi/klipper_logs/klippy.log"}}`

emtrax-ltd, 2022-07-30:
OK... the output looks good. Which url did you enter in the configuration "Connect Moonraker" under "Connection" -> "Address (URL)" .

CrownFlash, 2022-07-30:
Address (URL]
"http://fluiddpi.local/#/"

API-Key
""

Power Device
""

Frontend
""

emtrax-ltd, 2022-07-30:
OK... please try "http://fluiddpi.local/" without "#/" at the end.

CrownFlash, 2022-07-30:
And it worked ! 
You sure know quite well your plugin :)

closed 2022-07-30 by CrownFlash

emtrax-ltd, 2022-07-30:
nice & happy printing! :)

CrownFlash, 2022-07-30:
Since you are here i have a small question. Do you planned to implement basic function into the monitor section of cura ?
like: move axis, home axis, heating.

And thanks for the help again :)

emtrax-ltd, 2022-07-30:
Yes, the plan exists and the monitor UI is already prepared for it. but currently I hardly have time to extend the code -> this is a lot of work.