emtrax-ltd/Cura2MoonrakerPlugin #53: Upgraded to 5.1..0 plugin error

fenixryan, 2022-09-07:
Could no load, please remove or re-install

Tried from market place and from github

emtrax-ltd, 2022-09-07:
please attach the cura.log

Navyjetfixer, 2022-09-12:
Received error connection timed out connecting to Moonraker. 
[cura.log](https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/files/9543817/cura.log)

emtrax-ltd, 2022-09-12:
> Received error connection timed out connecting to Moonraker. [cura.log](https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/files/9543817/cura.log)

 HTTP 401: Unauthorized

Looks like missing/incorrect API key or unconfigured host address in [Authorization] -> trusted_clients

closed 2022-09-28 by emtrax-ltd

fenixryan, 2023-01-26:
Sorry for delay, sorted now. Multiple versions of cura merging manual install with plugin version causing issue. Un installed everything. Put only one version back and restored backup. Found the duplicate directory and nuked it. All good now.