emtrax-ltd/Cura2MoonrakerPlugin #55: Object of type SettingFunction is not JSON serializable

hapklaar, 2022-09-30:
Getting this error when using your plugin version 1.6.3 on Cura 5.2.0 Beta to upload to printer. Save to gcode works fine.

emtrax-ltd, 2022-10-01:
I can not reproduce your error with cura 5.2-beta - please attach your cura.log

Piezoid, 2022-10-01:
This error is triggered by a profile with a setting reset to its computed value: change a setting vlue then reset it with the `fx` icon.
It is not caused by Cura2Moonraker but UFPWriter. Saving to .ufp will crash Cura, even without Cura2Moonraker installed.

I've pinged the authors on [the PR that introduced it](https://github.com/Ultimaker/Cura/pull/13209#issuecomment-1251457665), but not response so far.

hapklaar, 2022-10-01:
> This error is triggered by a profile with a setting reset to its computed value: change a setting vlue then reset it with the `fx` icon. It is not caused by Cura2Moonraker but UFPWriter. Saving to [...]

That sounds plausible. The relevant log is as follows:

```
2022-10-01 13:05:36,002 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [969]: Backend finished slicing. Resetting process and socket.
2022-10-01 13:05:41,084 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [118]: 'info/send_slice_info' is turned off.
2022-10-01 13:05:41,317 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Exception: Unable to write to file CFFFP_Top_00: Object of type SettingFunction is not JSON serializable
2022-10-01 13:05:41,324 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Traceback (most recent call last):
2022-10-01 13:05:41,324 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Qt\Bindings\OutputDeviceManagerProxy.py", line 150, in _writeToDevice
2022-10-01 13:05:41,324 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     device.requestWrite(nodes, file_name, limit_mimetypes, file_handler, **kwargs)
2022-10-01 13:05:41,325 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Users\user\AppData\Roaming\cura\5.2\plugins\MoonrakerConnection\MoonrakerConnection\MoonrakerOutputDevice.py", line 94, in requestWrite
2022-10-01 13:05:41,325 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     if not meshWriter.write(self._stream, None):
2022-10-01 13:05:41,325 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "cura\Utils\Threading.py", line 31, in _call_on_qt_thread_wrapper
2022-10-01 13:05:41,326 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     return func(*args, **kwargs)
2022-10-01 13:05:41,326 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Program Files\Ultimaker Cura 5.2.0-beta\share\cura\plugins\UFPWriter\UFPWriter.py", line 87, in write
2022-10-01 13:05:41,326 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     json.dump(self._getSliceMetadata(), setting_textio, separators=(", ", ": "), indent=4)
2022-10-01 13:05:41,327 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\__init__.py", line 179, in dump
2022-10-01 13:05:41,327 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\encoder.py", line 431, in _iterencode
2022-10-01 13:05:41,327 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\encoder.py", line 405, in _iterencode_dict
2022-10-01 13:05:41,328 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\encoder.py", line 405, in _iterencode_dict
2022-10-01 13:05:41,328 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\encoder.py", line 405, in _iterencode_dict
2022-10-01 13:05:41,328 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\encoder.py", line 438, in _iterencode
2022-10-01 13:05:41,329 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\encoder.py", line 179, in default
2022-10-01 13:05:41,329 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: TypeError: Object of type SettingFunction is not JSON serializable
2022-10-01 13:05:49,245 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.006695985794067383 seconds
```

PhilBaz, 2022-10-01:
same here!
[image]
[json problem.zip](https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/files/9691137/json.problem.zip)
[cura.log](https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/files/9691138/cura.log)

emtrax-ltd, 2022-10-02:
There was a change/commit within UFPWriter causing this error -> https://github.com/Ultimaker/Cura/commit/4e785c3b61911bd689f63a7e1b8b63f3f443d127

Piezoid, 2022-10-02:
> There was a change/commit within UFPWriter which seems to fix this error -> [Ultimaker/Cura@4e785c3](https://github.com/Ultimaker/Cura/commit/4e785c3b61911bd689f63a7e1b8b63f3f443d127)

Have you tried with a computed setting? For example try setting an arbitrary value for "Minimum Wall Line Width", then reset by clicking the [image] icon.

The `master` branch and the `5.2.0-beta` tag presents this issue IME. I can't really try the released AppImage since it has some missing shared objects.

hapklaar, 2022-10-02:
> Have you tried with a computed setting? For example try setting an arbitrary value for "Minimum Wall Line Width", then reset by clicking the [image] icon.
>
> The `master` branch and the `5.2.0-beta` [...]

I've tried setting all values to their computed value in one of my profiles, but it makes no difference. Error remains.

Piezoid, 2022-10-02:
> I've tried setting all values to their computed value in one of my profiles, but it makes no difference. Error remains.

You'll have to use a stock "machine profile", not a user created profile. And then click "Discard current changes" in the profile menu.

hapklaar, 2022-10-02:
> You'll have to use a stock "machine profile", not a user created profile. And then click "Discard current changes" in the profile menu.

Yup, that works.

PhilBaz, 2022-10-02:
Sorry, Im not following. Is this a bug? Is there a fix?

PhilBaz, 2022-10-02:
Okay, if I use one of these profiles it works.....
[image]

PhilBaz, 2022-10-02:
@emtrax-ltd 

I replaced this .py with the linked version. problem persists.

emtrax-ltd, 2022-10-02:
Ok... can reproduce it now. It seems to be a bug within Cura (UFP Writer) itself. I think there is something wrong with the handling of "older" custom/inherited profiles - not all these profiles are causing this error.

PhilBaz, 2022-10-03:
@emtrax-ltd 

I just copied the UFPwriter folder from Cura 5.1 to 5.2 beta and it seems to be working fine now. 

Is this a bad idea?

C:\Program Files\Ultimaker Cura 5.2.0-beta\share\cura\plugins

curather, 2022-10-03:
I tested this but I don't think it is using your settings, my speed and temps were not correct in the g code.

> I just copied the UFPwriter folder from Cura 5.1 to 5.2 beta and it seems to be working fine, for now.
> 
> Is this a bad idea?
> 
> C:\Program Files\Ultimaker Cura 5.2.0-beta\share\cura\plugins

Piezoid, 2022-10-03:
> @emtrax-ltd
> 
> I just copied the UFPwriter folder from Cura 5.1 to 5.2 beta and it seems to be working fine, for now.
> 
> Is this a bad idea?

In general yes, but it might work in this case since there is only a few changes in this file between 5.1.1 and 5.2.0-beta.
You could also use my patched version: [UFPWriter.py](https://github.com/Piezoid/Cura/blob/fix-ufpwriter/plugins/UFPWriter/UFPWriter.py)

curather, 2022-10-03:
@Piezoid Thanks for this patch.  It seems to fix the issue.  You can also rebuild the custom profiles and that seems to fix it as well.

> In general yes, but it might work in this case since there is only a few changes in this file between 5.1.1 and 5.2.0-beta.
> You could also use my patched version: [UFPWriter.py](https://github.com/Piezoid/Cura/blob/fix-ufpwriter/plugins/UFPWriter/UFPWriter.py?rgh-link-date=2022-10-03T16%3A27%3A55Z)

PhilBaz, 2022-10-03:
@Piezoid @emtrax-ltd  Looks good! Thanks guys.

closed 2022-10-09 by emtrax-ltd

goldjunge91, 2023-02-17:
hi I get this error message but the funny part is that I get it only on some profiles for my printer, and all of my configs are started from the same profile 😩 
[image]