emtrax-ltd/Cura2MoonrakerPlugin #56: Timeout-all the time

andycrane, 2022-10-05:
Installed plugin in Cura 5.0.0 on a Windows 11 system with plenty of resources. Configured as instructions. Uploaded successfully on the first try, then failed to upload with a timeout error on each subsequent occasion. Uninstalled plugin, rebooted, checked IP allocation, reinstalled, reconfigured and it continues to time out each time. Thanks for the work on this plugin.

emtrax-ltd, 2022-10-09:
Please attach the cura.log? Is there any power device configured?

thghca, 2022-10-26:
When printer power turned off upload always timed out. No matter if I filled power device field or not.
Why does the plugin bother checking for power or klipper errors if I just want to upload a file without running job?

emtrax-ltd, 2022-10-30:
I will rework the upload phase in one of the next releases

kted, 2023-03-16:
Any progress on this?

closed 2023-08-10 by emtrax-ltd