emtrax-ltd/Cura2MoonrakerPlugin #58: Featurerequest: Upload to specific folder

webkonto, 2022-10-16:
I use two printers with the same host. Uploading a file would be easier, if I could specify a default "basic folder" for each printer. Example: I defined a folder "Voron24" and a folder "Anycubic" on my host using Fluidd. I'd like to have all models done with the Cura pinter name "MyVoron24" (which is my name for the Voron24 Cura printer setup) to the default folder "Voron24" on the remote host. This is because Moonraker does not distinguish between sepearte upload folders afaik (I can see all models in the same folder). That would be a great help since you can mix up model sliced for another printer easily.!

emtrax-ltd, 2022-10-25:
Questions:
1) Two printers defined in Cura, both using the same instance of moonraker/endpoint?
2) Are these separate subfolders under ~/.gcode_files/? (example please!)

Tompudex, 2022-10-29:
> 

example : 
first printer in 192.168.1.2:7125
secund 192.168.1.2:7126
first pach /home/pi/firstprinter_data/gcode
secund is /home/pi/secundprinter_data/gcode
the  shitty moonraker make error if we make 1 gcode folder the two printer. :/

webkonto, 2022-10-29:
OK, I have to check, maybe I am wrong. You may close the issue.

closed 2022-10-29 by webkonto