emtrax-ltd/Cura2MoonrakerPlugin #59: UFP

Roho81, 2022-10-24:
upload with UFP was working, then when I tried uploading a file, it would pause for a moment and i would get the following error:
"object of type SettingFunction is not JSON serializable"

if i change it to Gcode it works fine. This issue just started happening last night.

Liechtenstein1, 2022-10-24:
Same issue here! In my case it started with Cura 5.2.1

Roho81, 2022-10-24:
no mine was working with cura 5.2.1

Liechtenstein1, 2022-10-25:
If you can solve it, can you share how did you solved it in here?

Somed17, 2022-10-26:
> It's a bug in Cura and its older/inherited print profiles -> #55

I don't quite understand where to insert the new file UFPWriter.py . I don't see such a script in any of the plugin folders

Piezoid, 2022-10-26:
Try to save the gcode as .ufp, if it fails, then this is a Cura bug and should be reported on their issue tracker.
A project file and log file (containing the python traceback) would help to diagnose and reproduce this behavior.

> I don't quite understand where to insert the new file UFPWriter.py . I don't see such a script in any of the plugin folders

UFPWriter.py is not a plugin file but a part of Cura python application found it's application folder. Replacing it might help, but it's not recommended to  manually patching the app with a file from a old version.

That fix was merged before the release of 5.2 (Ultimaker/Cura#13473), but something could have been missed with the initial patch or with the changes they added for displaying the setting value on their cloud printing services.

Somed17, 2022-10-26:
> Try to save the gcode as .ufp, if it fails, then this is a Cura bug and should be reported on their issue tracker. A project file and log file (containing the python traceback) would help to diagnose and reproduce this behavior.

Yes, I couldn't save it as UFP, kura gives a critical error. If you change your personal profile to a standard one, then everything works

emtrax-ltd, 2022-10-27:
workarounds...
1) recreate your "causing" custom print profiles
2) wait for final release of cura 5.2.1  ;)

blendfish, 2022-11-13:
Just got the same with 5.2.1 after duplicating the built-in profiles and customising.

closed 2023-01-26 by emtrax-ltd