emtrax-ltd/Cura2MoonrakerPlugin #62: Cura 5.2.1: object of type settingfunction is not json serializable

phil2sat, 2022-10-30:
when i upgrade cura i get:
object of type settingfunction is not json serializable

in old version all is fine

`2022-10-30 17:11:17,530 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Exception: Unable to write to file CFFFP_Peppa_Wutz: Object of type SettingFunction is not JSON serializable
2022-10-30 17:11:17,531 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Traceback (most recent call last):
2022-10-30 17:11:17,532 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Qt\Bindings\OutputDeviceManagerProxy.py", line 150, in _writeToDevice
2022-10-30 17:11:17,532 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     device.requestWrite(nodes, file_name, limit_mimetypes, file_handler, **kwargs)
2022-10-30 17:11:17,533 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Users\phil2sat\AppData\Roaming\cura\5.2\plugins\MoonrakerConnection\MoonrakerConnection\MoonrakerOutputDevice.py", line 94, in requestWrite
2022-10-30 17:11:17,533 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     if not meshWriter.write(self._stream, None):
2022-10-30 17:11:17,534 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "cura\Utils\Threading.py", line 31, in _call_on_qt_thread_wrapper
2022-10-30 17:11:17,534 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     return func(*args, **kwargs)
2022-10-30 17:11:17,535 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Program Files\Ultimaker Cura 5.2.1\share\cura\plugins\UFPWriter\UFPWriter.py", line 88, in write
2022-10-30 17:11:17,535 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     json.dump(self._getSliceMetadata(), setting_textio, separators=(", ", ": "), indent=4)
2022-10-30 17:11:17,536 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\__init__.py", line 179, in dump
2022-10-30 17:11:17,536 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\encoder.py", line 431, in _iterencode
2022-10-30 17:11:17,537 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\encoder.py", line 405, in _iterencode_dict
2022-10-30 17:11:17,537 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\encoder.py", line 405, in _iterencode_dict
2022-10-30 17:11:17,538 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\encoder.py", line 405, in _iterencode_dict
2022-10-30 17:11:17,538 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\encoder.py", line 438, in _iterencode
2022-10-30 17:11:17,539 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "json\encoder.py", line 179, in default
2022-10-30 17:11:17,539 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: TypeError: Object of type SettingFunction is not JSON serializable`

emtrax-ltd, 2022-10-30:
It's a bug in Cura 5.2.1 and its older/inherited print profiles -> #55 & #59

phil2sat, 2022-10-31:
Recreated my Cura Profile, issue is fixed

jonny7737, 2022-11-28:
When will fix be released?  5.2.1 still has the issue.

phil2sat, 2022-11-29:
> When will fix be released? 5.2.1 still has the issue.

recreate cura profile, this is the fix