emtrax-ltd/Cura2MoonrakerPlugin #64: Cure 5.2.x on macOs uses a other Plugin-Directory

daniel-volz, 2022-11-13:
Hello,

It seems, that in the current Cura 5.2, a different plugin directory is used.
In Cura 5.x or newer, the default installation path is Ultimaker Cura.app -> Contents -> Resources -> share -> cura -> plugins.

But this Plugin-Directory seems to be for bundled Plugins only.
Cura can´t find additional Plugins here.

I had to put the plugin in the directory:
/Library/Application Support/cura/5.2/plugins
to get it to work.

There is a readme file in the old plugins directory that points this out.
Maybe the correct directory will be added here in the readme from Cura 5.2

Greetings, Daniel

ward1975, 2022-12-18:
hi i tried and tried uploading to my sonic pad and it dose not work its coming up with error all the time every think is connected but its not loading up to it i put my ip address in and it dose not work ?
[image]

pedrolamas, 2023-01-14:
Wouldn't it be easier to just tell users to click "Help" -> "Show Configuration Folder" and then open the "plugins" subfolder?