emtrax-ltd/Cura2MoonrakerPlugin #65: Object of type is not JSON serializable - Error during upload

Darkmoner, 2023-01-08:
Hello,

After clicking Upload to Printer below error apears: [Object of type is not JSON serializable]

[image]

I'm sure it is releated to Cura profile, because after changing to Standard - default , everything works fine.

[bad_profile.zip](https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/files/10369634/bad_profile.zip)

Here is my profile what makes crash

atroph, 2023-01-11:
I've noticed that UFP w/ thumbnail is what is causing the problem for me. 

Standard upload "gcode" format works fine.

atroph, 2023-01-11:
Looks like a cura bug. 

See here. https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/issues/55#issuecomment-1265732978

emtrax-ltd, 2023-02-18:
https://github.com/Ultimaker/Cura/issues/13675

closed 2023-02-18 by emtrax-ltd