emtrax-ltd/Cura2MoonrakerPlugin #68: JSON error when upload

darxfame, 2023-02-09:
Hello. I'm download CURA 5.2.2 and slice my model with tumbnail setting.
When i'm upload it to moonraker i'm had json error.

[image]

simeonov, 2023-02-15:
@darxfame did you manage to fix or found solution to this? I have exact same issue with 5.2.2. It was working fine. I just restart my computer and then stopped working and it shows the error.

emtrax-ltd, 2023-02-18:
https://github.com/Ultimaker/Cura/issues/13675

iqubik, 2023-02-27:
same issue

Aruta79, 2023-03-28:
I found a possible workaround, which may also point at the source of the issue. For me, it shows the error about 95% of the time when trying to upload models with various changes in the profile, but if I change the main material temperature to a custom one directly within the profile, it uploads 100% of the time. I just have to change the temp by 1deg. It seems to be that it's some conflict either with the way the profile is saved, or with another plugin which exposes settings in the profile.

fieldOfView, 2023-04-10:
This should be fixed in Cura 5.3

closed 2023-08-10 by emtrax-ltd