emtrax-ltd/Cura2MoonrakerPlugin #70: moonraker cura plugin  conflicting with  octoprint cura plugin

Arketsu, 2023-04-09:
Currently I have 2 printers.
1. Ender 3 S1 Running with octoprint.
2. Ender 3 V2 running with Klipper , Moonraker.

Recently installed Moonraker plugin to be able to send prints from cura to my klipper printer but found out that the moonraker plugin is conflicting with octoprint plugin. how?

The monitor tab in cura does not show anything if the moonraker plugin is installed when the octoprint plugin is already installed. Only work around I found was to add the url of my moonraker instance and the urls of the camera on my Ender 3 S1 which is the one running octoprint then I can see everything normal in the monitors tab. If I do this another problem pops up which is that cura will default to moonraker and I have to change the destination of the print every time otherwise sends it to the moonraker instance instead of the octoprint.

PS: I will submit the same thing to the octoprint cura plugin github.

Arketsu, 2023-04-14:
@emtrax-ltd  do you think this is an easy fix?

emtrax-ltd, 2023-04-14:
> @emtrax-ltd do you think this is an easy fix?

I'll check it out and fix it if possible.

emtrax-ltd, 2023-04-17:
give it a try -> [v1.7.0](https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/releases/tag/v1.7.0)

Arketsu, 2023-04-17:
so far so good!! thanks you!!

closed 2023-04-17 by Arketsu