emtrax-ltd/Cura2MoonrakerPlugin #73: Select target for upload

ThreeD-Michael, 2023-05-06:
First of all: thanks for this plugin!

I have 2 completely identical printers (with different Moonraker IP addresses). I am only using 1 Cura printer profile for these 2 printers and I would like to be able to upload a gcode file to any one of these 2 printers. Currently I have to save the sliced file to disk and upload them manually with the Fluid web interface. A menu to configure and select multiple Moonraker instances in Cura would be helpful.

Unfortunately, I'm not familiar with Python and Curas plugin management, otherwise I would just try to create a second instance of your plugin with a different name.

simeonov, 2023-06-01:
Hey @ThreeD-Michael I have exactly the same problem. It will be nice to have the option to choose between multiple connections.

emtrax-ltd, 2023-06-01:
You have to create each printer (even identical ones) separately in Cura and then connect/configure them with the desired output plugin. This is the concept of Cura.

closed 2023-08-10 by emtrax-ltd