emtrax-ltd/Cura2MoonrakerPlugin #81: Uploading fails if klipper not running

wollew, 2023-12-25:
Uploading files to moonraker fails if the printer/klipper is not connected:
`2023-12-25 12:14:48,961 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [369]: request[31c2ee3f][get][PyQt6.QtCore.QUrl('http://192.168.1.22:7125/printer/info')][timeout=None][no-data] got an QNetworkReplyError Error transferring http://192.168.1.22:7125/printer/info - server replied: Klippy Host not connected. The server returned: b'{"error": {"code": 503, "message": "Klippy Host not connected", "traceback": "Traceback (most recent call last):\\n\\n  File \\"/home/pi/moonraker/moonraker/app.py\\", line 641, in _process_http_request\\n    result = await self.api_defintion.request(\\n\\n  File \\"/home/pi/moonraker/moonraker/klippy_connection.py\\", line 558, in request\\n    raise ServerError(\\"Klippy Host not connected\\", 503)\\n\\nmoonraker.utils.ServerError: Klippy Host not connected\\n\\n\\nThe above exception was the direct cause of the following exception:\\n\\n\\nTraceback (most recent call last):\\n\\n  File \\"/home/pi/moonraker-env/lib/python3.9/site-packages/tornado/web.py\\", line 1713, in _execute\\n    result = await result\\n\\n  File \\"/home/pi/moonraker/moonraker/app.py\\", line 624, in get\\n    await self._process_http_request(RequestType.GET)\\n\\n  File \\"/home/pi/moonraker/moonraker/app.py\\", line 645, in _process_http_request\\n    raise tornado.web.HTTPError(\\n\\ntornado.web.HTTPError: HTTP 503: Klippy Host not connected\\n"}}'`

My printer is not usually turned on all the time but I would still like to be able to slice and upload files to moonraker. This works through the Mainsail/moonraker web frontend, but the Cura plugin fails because it requests /printer/info which (correctly) responds with 503 because printing would not be possible. I know I could configure moonraker's power device in the plugin which then fixes things but I don't really see why my printer needs to be turned on just to upload files.

scabana, 2024-01-03:
I got the same kind of problem. I believe those lines of code https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/blob/8faac6255e49c50646f1b8b3eb1d9f6547719a73/MoonrakerConnection/MoonrakerOutputDevice.py#L262-L266 are the issue. I have setup: 

```
[file_manager]
queue_gcode_uploads: True

[job_queue]
load_on_startup: True
```
in `moonraker.conf` + `on_when_job_queued: True` on a power device. I believe that if the current status check would be removed, the printer would power up. After having issues, I re-configured in the extension here the power devices, but I'd rather disable the check in the plugin and let klipper's config do the power on of the printer and not have the power on config in 2 places. I guess a checkbox of some kind near the power devices option to disable the check would be the right thing to do.

emtrax-ltd, 2024-01-04:
If the printer is off, deselect the "Start print job" option in the upload dialog. Job queuing is not yet supported by the plugin.

wollew, 2024-01-04:
I am almost 100% positive that I hadn't selected "Start print job" but uploading did not work anyway. Currently the printer is on (and printing), I'll try again once it's done and turned off.

scabana, 2024-01-04:
@emtrax-ltd , it's not as simple as I said (remove the check) because of how the code is laid out. I'm far from a python pro but I did clone the repo and pointed the plugin to it and with those two changes: 

[image]

and 

[image]

(remember, big hack to get the code invoked ;))

The plugin did trigger the print with the printer being offline. The gcode file was queued and my power devices (setup in moonraker) did turn on and the print did start.

@wollew, reading your description again, I think it's not the exact same scenario as the one I explained.

wollew, 2024-01-05:
Ok, I tried again, "Start print job" was definitely not selected, this was the result:
[image]

emtrax-ltd, 2024-01-05:
Try this branch: https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/tree/pwrdevfix - there are many changes in MoonrakerOutputDevice.py now.

scabana, 2024-01-05:
Wait, upload without print did work.

emtrax-ltd, 2024-01-05:
> Wait, upload without print did work.

Important: the “Start print job” option in the upload dialog requires either an active printer or a named power device that activates the printer. In your configuration it only works without "Start print job" because no power device has been named - the Moonraker option "queue_gcode_uploads: True" ensures that a print job is created and, if desired, the option "on_when_job_queued: True" for the power device ensures that the printer is activated by creating the job.

wollew, 2024-01-07:
I tried the branch, but uploading still fails:

```
Uploading to Moonraker at http://octoprint.at.home:7125/ was not successful.

The status of the printer is 'error'.

mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer - Unknown error
```

No exceptions in the log file.

emtrax-ltd, 2024-01-07:
> I tried the branch, but uploading still fails:
>
> ```
> Uploading to Moonraker at http://octoprint.at.home:7125/ was not successful.
>
> The status of the printer is 'error'.
>
> mcu 'mcu': Unable to connect [...]

it looks like the upload implies a "start print job". Do you use the upload dialog or the "fire & forget" mode?

wollew, 2024-01-07:
> Do you use the upload dialog or the "fire & forget" mode?

I am using the upload dialog, "Start print job" is unchecked.

emtrax-ltd, 2024-01-08:
> > Do you use the upload dialog or the "fire & forget" mode?
> 
> I am using the upload dialog, "Start print job" is unchecked.

For further investigation of this problem the cura.log is required

wollew, 2024-01-10:
Sorry for taking a little bit longer, this morning I tried again, interesting enough I got a different error message:

```
Uploading to Moonraker at http://octoprint.at.home:7125/ was not successful.

NetworkError.ServiceUnavailableError - Error transferring http://octoprint.at.home:7125/printer/info - server replied: Klippy Host not connected
```

Relevant part of the  [cura.log](https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/files/13884277/cura.log) is attached.

closed 2024-05-09 by emtrax-ltd

wollew, 2024-05-10:
Unfortunately this does not work for me, /printer/info returns 503 if the printer is not turned on and so we end up in the else branch in  `_checkPrinterStatus`. I can try to create a PR if you're interested.

emtrax-ltd, 2024-05-10:
> Uploading to Moonraker at http://octoprint.at.home:7125/ was not successful.
> 
> NetworkError.ServiceUnavailableError - Error transferring http://octoprint.at.home:7125/printer/info - server replied: Klippy Host not connected

It fails because the Klipper service is not active/started on the Pi. This service should actually be active all the time, even if the printer itself is not switched on or the printer's MCU/board is without power.

wollew, 2024-05-10:
> It fails because the Klipper service is not active/started on the Pi. This service should actually be active all the time, even if the printer itself is not switched on or the printer's MCU/board is [...]

It is true that the klipper service is not running, it automatically ends/dies when I turn off my Ender running klipper (because it does have a USB connection any more). And it automatically restarts (via systemd) as soon I power up the printer. Never had any problems with that and frankly I thought it is the default behaviour. So klipper.service should be up when the MCU is powered down? 
But even if klipper is not ready, moonraker sure is (which can be queried via /server/info). So I don't really see why uploads shouldn't be possible.

emtrax-ltd, 2024-05-10:
> Try this branch: https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/tree/pwrdevfix

> It is true that the klipper service is not running, it automatically ends/dies when I turn off my Ender running klipper (because it does have a USB connection any more). And it automatically [...]

Try latest version of branch: https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/tree/pwrdevfix - there is a new commit!

wollew, 2024-05-10:
LGTM, will confirm once my currently running print is done. Thank you.

wollew, 2024-05-10:
As expected, this fixed the problem for me.