emtrax-ltd/Cura2MoonrakerPlugin #82: Uploading fails (timeout)

Nuc1eoN, 2023-12-25:
Hello I have just installed `Cura2MoonrakerPlugin` and configured my priters IP including port number.

However uploading fails:
[image]

The mailsail interface is reachable over my web browser. Moonraker also is up and running:
http://192.168.0.107:7125/server/info
```json
{"result": {"klippy_connected": true, "klippy_state": "ready", "components": ["klippy_connection", "application", "jsonrpc", "internal_transport", "websockets", "dbus_manager", "database", "file_manager", "klippy_apis", "secrets", "template", "shell_command", "machine", "data_store", "proc_stats", "job_state", "job_queue", "http_client", "announcements", "webcam", "extensions", "history", "update_manager", "octoprint_compat", "authorization"], "failed_components": [], "registered_directories": ["config", "logs", "gcodes", "config_examples", "docs"], "warnings": [], "websocket_count": 2, "moonraker_version": "v0.8.0-231-gcc5dea7", "missing_klippy_requirements": [], "api_version": [1, 4, 0], "api_version_string": "1.4.0"}}
```

[image]

I have also checked firewall but it all seems ok. Only the Cura2MoonrakerPlugin seems to have a problem connecting.

Cura v5.6

ThGIGA, 2024-01-04:
The same thing is happening to me.  Please help.  !!!

I'm using the OctoPrint plugin at the moment which "does work" (kinda), but it's missing information when I see Mainsail.  

However, I would prefer to use the Moonraker plugin because when i upload the file to the printer because it appears with more information un the web.

Nuc1eoN, 2024-01-04:
@ThGIGA just for your information, orcaslicer works great with mainsail out of the box, no need for a plugin!

emtrax-ltd, 2024-01-04:
Are you using the latest version 1.7.2 of the plugin?

ThGIGA, 2024-01-05:
@emtrax-ltd, yes ,I have downloaded it from the market and from here, Github. And I haven't been able to get it to work.

Nuc1eoN, 2024-01-06:
@emtrax-ltd I am on v1.7.2 from marketplace, yes.

emtrax-ltd, 2024-01-07:
Try out this branch: https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/tree/pwrdevfix - in this branch are some changes in the process of connecting and uploading

ThGIGA, 2024-01-09:
[image]
 
 Still not working ... Sad

emtrax-ltd, 2024-01-09:
> ![Sin título](https://private-user-images.githubusercontent.com/84819548/295078674-aa284b82-260d-4861-845d-1347542b4b17.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ3OTYzMDgsIm5iZiI6MTcwNDc5NjAwOCwicGF0aCI6Ii84NDgxOTU0OC8yOTUwNzg2NzQtYWEyODRiODItMjYwZC00ODYxLTg0NWQtMTM0NzU0MmI0YjE3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA5VDEwMjY0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYzMmUzMTllYjQzOWVmMTcxNzY0ZTU1YjJlMDA1ZWU5YzE5MTY1NGYwMDBkNTdhN2M1MTI0NjU3ZTM2MDY2MDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.jUnXiaxf6CVK_lII1z7rBs2o7G5n0eTzETRtQGs3wqM)
> 
> Still not working ... Sad

I don't know about your Moonraker/Klipper installation, but it looks like the Moonraker API is not reverse-proxyed (Nginx) over port 80. Therefore, it is probably necessary to specifically specify the port of the Moonraker API in the plugin configuration -> http://192.168.0.127:7125

ThGIGA, 2024-01-10:
@emtrax-ltd 
It seems to be working now with the latest version and setting the IP with the port as shown in the image of the plugin

@Nuc1eoN Try this with the latest version, see if it works for you too. 
[image]
[image]
[image]

Nuc1eoN, 2024-01-11:
It does not work here:
[image]

>I don't know about your Moonraker/Klipper installation, but it looks like the Moonraker API is not reverse-proxyed (Nginx) over port 80. Therefore, it is probably necessary to specifically specify the port of the Moonraker API in the plugin configuration -> http://192.168.0.127:7125/

I have the stock mainsail OS setup on raspi pretty much.

ThGIGA, 2024-01-13:
@Nuc1eoN did u try on this branch: https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/tree/pwrdevfix ?

Nuc1eoN, 2024-01-13:
Yes that is the branch I've used.

Before I've used that branch, the error message was different (timeout). Now it says 'NetworkError...' like in the screenshot

closed 2024-05-11 by emtrax-ltd