emtrax-ltd/Cura2MoonrakerPlugin #84: Default subfolder

Bart1909, 2024-01-05:
Hi,

currently it is already possible to upload files in a subfolder e.g. with 'sda/myfile.gcode'

It would be nice, if there will be a configuration setting to set a default folder, which will always be prefixed at the file names.

Best regards

Tobi

reinew, 2024-01-06:
Had the same thought, would be nice. Even better would be to get a dropdown menu with all subfolders that exist on the printer and that the plugin remembers the last folder that is chosen.

ManeLippert, 2024-01-21:
As starting point there exists already an plugin for Octoprint which lets you save in any folder before upload.

- [Printjob Naming Plugin](https://github.com/fieldOfView/Cura-CustomJobPrefix)
- [Octoprint Plugin](https://github.com/fieldOfView/Cura-OctoPrintPlugin)

reinew, 2024-01-21:
> As starting point there exists already an plugin for Octoprint which lets you save in any folder before upload.
> 
> 
> 
> - [Printjob Naming Plugin](https://github.com/fieldOfView/Cura-CustomJobPrefix)
> 
> - [Octoprint Plugin](https://github.com/fieldOfView/Cura-OctoPrintPlugin)

For thos who don't use Octoprint, that plugin will not help. I use Klipper with Mainsail frontend.

ManeLippert, 2024-01-21:
> For thos who don't use Octoprint, that plugin will not help. I use Klipper with Mainsail frontend.

@reinew it was meant as starting point for development of the feature not as recommendation for usage. Apologise for the misunderstanding.

jaredmullins08, 2024-03-17:
Agree. The printer default folder has such limited storage space. Please, if someone could code it to allow a default file path at the beginning of each file name to save it to the USB folder which has potential to hold much more. You currently cannot find and print the file from certain printer's control panel unless saved in a subfolder.

emtrax-ltd, 2024-05-09:
Try this branch: https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/tree/pthuplext - there are now many changes for configurable upload pathes (default/list).

emtrax-ltd, 2024-05-16:
Release 1.9.1 is now submitted to the Cura marketplace and we have to wait for the QC process.

emtrax-ltd, 2024-05-21:
Release 1.9.1 is now available in the in the Cura marketplace.

closed 2024-05-21 by emtrax-ltd