emtrax-ltd/Cura2MoonrakerPlugin #85: Getting internal server error in Cura when trying to upload ufp

hapklaar, 2024-03-09:
Not sure if this is a plugin issue, or maybe moonraker or something specific to my config...

Often, but not always, I'm greeted by an internal server error message in Cura, when I try to upload a file. Retry helps most of the time, or I restart Moonraker and the retry a couple more times. Eventually it will go through.

In `moonraker.log` I see:
```
2024-03-09 10:39:24,563 [application.py:prepare()] - Upload Request Received from 192.168.1.11
Content-Type: multipart/form-data; boundary=boundary_.oOo._CHJ1WwmYFcomPr2mxQwcPg8fMnMq/RoM
2024-03-09 10:39:24,572 [application.py:data_received()] - Chunk Parsing Error
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/application.py", line 974, in data_received
    await evt_loop.run_in_thread(self._parser.data_received, chunk)
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/streaming_form_data/parser.py", line 82, in data_received
    raise ParseFailedException(
streaming_form_data.parser.ParseFailedException: _parser.data_received failed with delimiting multipart stream into parts
2024-03-09 10:39:24,627 [web.py:log_exception()] - 500 POST /server/files/upload (192.168.1.11): File Upload Parsing Failed
2024-03-09 10:39:24,633 [application.py:log_request()] - 500 POST /server/files/upload (192.168.1.11) [_TRUSTED_USER_] 70.86ms
```
`cura.log` for the same event:
```
2024-03-09 11:39:24,970 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onUploadFilenameAccepted [238]: FileName set to CFFFP_Collar.ufp.
2024-03-09 11:39:24,971 - INFO - [MainThread] MoonrakerConnection.MoonrakerSettings.saveConfig [35]: MoonrakerSettings save config for printer... id:Rattatat
2024-03-09 11:39:24,971 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onUploadFilenameAccepted [250]: StartPrint set to True.
2024-03-09 11:39:24,972 - INFO - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onUploadFilenameAccepted [254]: Connecting to Moonraker at http://192.168.1.40/.
2024-03-09 11:39:24,972 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._getPowerDeviceStatus [274]: Checking printer device [power RatRig] status.
2024-03-09 11:39:25,314 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._checkPowerDeviceStatus [288]: Power device [power RatRig] status == 'on'; startPrint is True => Calling _getPrinterStatus()
2024-03-09 11:39:25,336 - INFO - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onPrinterOnline [348]: Uploading file 'CFFFP_Collar.ufp' [format: ufp].
2024-03-09 11:39:25,413 - ERROR - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onRequestError [485]: <NetworkError.InternalServerError: 401>
2024-03-09 11:39:25,414 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._resetState [201]: Reset state of device.
```

Here I retried and it works:
`moonraker.log`
```
2024-03-09 10:39:38,216 [application.py:prepare()] - Upload Request Received from 192.168.1.11
Content-Type: multipart/form-data; boundary=boundary_.oOo._4O3KYFMsp0EBjZEklOJ2GyIUaBu8pJlw
2024-03-09 10:39:38,416 [application.py:post()] - Processing Uploaded File: /CFFFP_Collar.ufp
2024-03-09 10:39:38,625 [shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is enabled

2024-03-09 10:39:38,912 [shell_command.py:pipe_data_received()] - INFO:prepropress_cancellation:Shapely not found, complex hulls disabled

2024-03-09 10:39:38,919 [shell_command.py:pipe_data_received()] - INFO:metadata:Performing Object Processing on file: CFFFP_Collar.gcode, sliced by Cura

2024-03-09 10:39:47,797 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/components/file_manager/metadata.py -p /home/pi/printer_data/gcodes -f "CFFFP_Collar.gcode" -u "/tmp/moonraker.upload-607707.mru" --check-objects) successfully finished
2024-03-09 10:39:47,814 [application.py:log_request()] - 201 POST /server/files/upload (192.168.1.11) [_TRUSTED_USER_] 9598.97ms
```
`cura.log`
```
2024-03-09 11:39:38,826 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onUploadFilenameAccepted [238]: FileName set to CFFFP_Collar.ufp.
2024-03-09 11:39:38,827 - INFO - [MainThread] MoonrakerConnection.MoonrakerSettings.saveConfig [35]: MoonrakerSettings save config for printer... id:Rattatat
2024-03-09 11:39:38,827 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onUploadFilenameAccepted [250]: StartPrint set to True.
2024-03-09 11:39:38,828 - INFO - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onUploadFilenameAccepted [254]: Connecting to Moonraker at http://192.168.1.40/.
2024-03-09 11:39:38,829 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._getPowerDeviceStatus [274]: Checking printer device [power RatRig] status.
2024-03-09 11:39:38,976 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._checkPowerDeviceStatus [288]: Power device [power RatRig] status == 'on'; startPrint is True => Calling _getPrinterStatus()
2024-03-09 11:39:38,985 - INFO - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onPrinterOnline [348]: Uploading file 'CFFFP_Collar.ufp' [format: ufp].
2024-03-09 11:39:48,595 - INFO - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onFileUploaded [396]: Upload completed.
2024-03-09 11:39:48,597 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._resetState [201]: Reset state of device.
```

any ideas?

fluxor99232, 2024-04-16:
Interestingly the quickest and most reliable way for me to get the file to upload to go through is to refresh the fluidd interface in the browser. Also making sure there's no existing file with the same name seems to be important.

emtrax-ltd, 2024-05-09:
Tra release v1.8.0

closed 2024-05-11 by emtrax-ltd