emtrax-ltd/Cura2MoonrakerPlugin #91: May update of moonraker

Jo01A1, 2024-06-08:
I’m fairly new to 3D printing so please assume complete ignorance.
I have an Ender 5 Plus with silent motherboard, Swiss micro NG with revo direct drive .
I have been using Moonraker with a creality sonic pad without any problems since last December. Since the latest update1.9.1 it is extremely difficult to use. Any attempt to send the stl file closeS the slicer without any error messages. If I try to send it to the creality cloud it sends an error report intermittently and/or closeS the slice program down. I can still use an SD  card to transfer the files but it’s very disappointing. Can you help? Is there a way of reverting to the previous version?
Thank you

emtrax-ltd, 2024-06-10:
cura version & cura.log?

Jo01A1, 2024-06-10:
Creality Slicer  which I’ve since found out is no longer supported or updated.
I will be switching to Ultimaker and hopefully it will start working again

closed 2024-06-12 by emtrax-ltd