emtrax-ltd/Cura2MoonrakerPlugin #96: Upload - Internal Server Error

mikepiko, 2024-09-27:
Hi, i often get "Internal Server Error" Messages when uploading files to my printer.

When i try it multiple times (like 2 times in this example of the Log below) it works.

`2024-09-27 15:47:16,977 [server.py:add_log_rollover_item()] - Klipper Version: v0.12.0-298-g0532a41c7-dirty
2024-09-27 15:47:18,545 [klippy_connection.py:_request_initial_subscriptions()] - Webhooks Subscribed
2024-09-27 15:47:18,549 [klippy_connection.py:_request_initial_subscriptions()] - GCode Output Subscribed
2024-09-27 15:47:18,796 [job_state.py:_handle_started()] - Job state initialized: standby
2024-09-27 15:47:19,069 [klippy_connection.py:_check_ready()] - Klippy ready
2024-09-27 15:47:19,548 [data_store.py:_init_sensors()] - Configuring available sensors: ['temperature_sensor EBB_NTC', 'heater_bed', 'temperature_fan PSU', 'temperature_fan RPI3', 'temperature_sensor Chamber', 'temperature_sensor BedMat', 'extruder']
2024-09-27 15:48:05,364 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 192.168.100.2
2024-09-27 15:48:05,409 [application.py:prepare()] - Upload Request Received from 192.168.100.2
Content-Type: multipart/form-data; boundary=boundary_.oOo._0lAUmfW4055nTwqIF/OOdQvKqa1ZV2xs
2024-09-27 15:48:05,418 [application.py:data_received()] - Chunk Parsing Error
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/application.py", line 975, in data_received
    await evt_loop.run_in_thread(self._parser.data_received, chunk)
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/streaming_form_data/parser.py", line 82, in data_received
    raise ParseFailedException(
streaming_form_data.parser.ParseFailedException: _parser.data_received failed with delimiting multipart stream into parts
2024-09-27 15:48:08,102 [web.py:log_exception()] - 500 POST /server/files/upload (192.168.100.2): File Upload Parsing Failed
2024-09-27 15:48:08,106 [application.py:log_request()] - 500 POST /server/files/upload (192.168.100.2) [_TRUSTED_USER_] 2698.60ms
2024-09-27 15:48:10,683 [application.py:prepare()] - Upload Request Received from 192.168.100.2
Content-Type: multipart/form-data; boundary=boundary_.oOo._w5qExr5YYASCStaKDI6F5h+zHxLmFJUz
2024-09-27 15:48:17,019 [application.py:post()] - Processing Uploaded File: /V350_Shark-Shark_v4_A.ufp
2024-09-27 15:48:17,450 [shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is disabled

2024-09-27 15:48:18,331 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/components/file_manager/metadata.py -p /home/pi/printer_data/gcodes -f "V350_Shark-Shark_v4_A.gcode" -u "/tmp/moonraker.upload-85.mru") successfully finished
2024-09-27 15:48:18,333 [klippy_apis.py:start_print()] - Requesting Job Start, filename = V350_Shark-Shark_v4_A.gcode`

any idea what could go wrong ?

I got Mainsail 2.12.0 with Moonraker v0.9.3-0-g71f9e677 on my Voron Printer.

Thx

Mike

kpbicka, 2024-10-30:
Same problem

mikepiko, 2024-11-02:
`2024-10-18 16:35:58,664 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onUploadFilenameAccepted [234]: FileName set to V350_BunnyRabbit__Presupported_v05R.ufp.
2024-10-18 16:35:58,664 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onUploadFilenameAccepted [246]: StartPrint set to True.
2024-10-18 16:35:58,665 - INFO - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onUploadFilenameAccepted [250]: Connecting to Moonraker at http://voron24.starfleetcommand.de/.
2024-10-18 16:35:58,697 - INFO - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onPrinterOnline [344]: Uploading file 'V350_BunnyRabbit__Presupported_v05R.ufp' [format: ufp].
2024-10-18 16:35:59,184 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [316]: SliceInfo sent successfully
2024-10-18 16:36:01,448 - ERROR - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._onRequestError [480]: <NetworkError.InternalServerError: 401>
2024-10-18 16:36:01,450 - DEBUG - [MainThread] MoonrakerConnection.MoonrakerOutputDevice._resetState [197]: Reset state of device.`

Here is a sample error out of the Cura.log

kfong257, 2025-01-26:
Same issue, using:
OrcaSlicer 2.2.0
moonraker v0.9.3-19-gd4d20744

from moonraker.log. 

2025-01-25 22:38:43,739 [application.py:prepare()] - Upload Request Received from 192.168.1.XX
Content-Type: multipart/form-data; boundary=------------------------7863bcd10f0cf948
2025-01-25 22:38:43,753 [application.py:data_received()] - Chunk Parsing Error
Traceback (most recent call last):
  File "/home/biqu/moonraker/moonraker/components/application.py", line 977, in data_received
    await evt_loop.run_in_thread(self._parser.data_received, chunk)
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/streaming_form_data/parser.py", line 82, in data_received
    raise ParseFailedException(
streaming_form_data.parser.ParseFailedException: _parser.data_received failed with delimiting multipart stream into parts
2025-01-25 22:38:45,320 [web.py:log_exception()] - 500 POST /api/files/local (192.168.1.XX): File Upload Parsing Failed
2025-01-25 22:38:45,324 [application.py:log_request()] - 500 POST /api/files/local (192.168.1.XX) [_TRUSTED_USER_] 1586.89ms