emtrax-ltd/Cura2MoonrakerPlugin #97: temp changes?

sleepjr, 2024-10-26:
first off im a newbie with roughly an hr of experience on klipper.
my problem im experiencing is when i send the code to fluidd it all looks good, preheats at the right temps (210/65) but as  soon as i tell fluid to print it it drops my temps down to 140. i don't know if this is something that im exporting wrong. any help would be appreciated

DemonChicken1111, 2024-10-26:
Also having this issue on my new Neptune 4 with klipper. Have my print settings in a custom material in Cura but every time I send it to the printer via Cura2Moonraker it tries to target 140 C. 

[image]

I have not tried to manually upload files to test if it is Cura or the plugin. Once my current print finishes I will run more troubleshooting to see if that is the case. 

**Steps to reproduce:
- Create custom material
- Set custom temperature 
- Slice any file
- Send to printer via Cura2Moonraker
- Check klipper or printer (target temp is always at 140, not custom value)

DemonChicken1111, 2024-10-27:
After trying a print via the USB port I can confirm that this is most likely a Cura issue. The problem persists regardless of whether the g-code is sent over Cura2Moonraker or not.